espressif/esp-idf-ci-action #13: How to build a project with a specific commit of esp-idf rather than a tag

alic-maker, 2022-03-09:
Hi everyone,

I want to compile my project with one of the commit of ESP-IDF. Now allow me to do so? What should I do if it can?

Thanks!

kumekay, 2022-03-09:
Hi @alic-maker 
This action uses docker images from https://hub.docker.com/repository/docker/espressif/idf/tags?page=1&ordering=last_updated under the hood. However, these images are not generated for every commit.

If you need to run a container with a particular IDF commit there are some options, but you have to take more steps.
For example, in your workflow, you can clone ESP-IDF, check out the right commit, then build the docker image in `tools/docker/Dockerfile` and run it similar to https://github.com/espressif/esp-idf-ci-action/blob/main/action.yml#L25

alic-maker, 2022-03-15:
Thank you, I implemented it in another way, here: https://github.com/espressif/esp-box/blob/master/.github/workflows/build.yml#L25

closed 2022-03-15 by alic-maker