espressif/esp-idf-ci-action #17: Cannot access build output from GitHub host runner

lazarofrancoe, 2022-05-25:
Hi,
I couldn't find a way to access the output file of my build from the GitHub host runner, as I need to upload it to AWS S3.

Should I mount a Docker volume (shared between the Esp Idf Docker image and the GitHub host runner) and run the Docker image manually without using this GitHub action at all?

My yml file is:
`name: production

on:
  push:
    branches: [main]

jobs:
  build:
    runs-on: ubuntu-latest
    permissions:
      id-token: write
      contents: read

    steps:
      - uses: actions/checkout@v2

      - name: Configure AWS Credentials
        uses: aws-actions/configure-aws-credentials@v1
        with:
          role-to-assume: arn:aws:iam::XXXXXXXXXXX:role/GitHubActionsRole
          aws-region: us-east-1

      - name: esp-idf build
        uses: espressif/esp-idf-ci-action@main
        with:
          esp_idf_version: v4.4
          command: idf.py build
          target: esp32

      - name: Push to AWS S3
        run: aws s3api put-object --bucket electronica-builds-production --key folder/output-file.bin --body /app/build/output-file.bin
`
Obviously, the aws command won't work because the /app/build/output-file.bin file refers to the Docker image directory.

Thanks in advance!

kumekay, 2022-05-26:
Hi @lazarofrancoe  

Try `${GITHUB_WORKSPACE}/build/output-file.bin`

lazarofrancoe, 2022-05-26:
Thanks Sergei,
 I could see the command that the action executes: 
`docker run -t -e IDF_TARGET="${IDF_TARGET}" -v "${GITHUB_WORKSPACE}:/app" -w "/app/" espressif/idf:v4.4 idf.py build
`
but I couldn't find the exact path to the shared folder.

Thanks for your support!

closed 2022-05-26 by lazarofrancoe