espressif/esp-idf-ci-action #21: esp-idf-ci-action doesn't integrate with the rest of the build system

xobs, 2022-07-22:
I'm trying to build a project that uses `esp-idf` for the primary build. This project additionally uses `python` and `nodejs` in order to generate and compress web pages.

I have actions in my build script that install these projects using the official actions, however these are not picked up by the `esp-idf` project, and the failure is somewhat opaque.

This action should integrate with the rest of the builders, and allow the usage of other steps in the build process.

kumekay, 2022-07-25:
Hi @xobs, 

`esp-idf-ci-action` runs its own docker container based on https://hub.docker.com/r/espressif/idf image. It cannot access any environment installed on the runner VM directly. 

If you need to use additional tools (like nodejs) for your build, you can either install ESP-IDF to your runner machine directly, following https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/linux-macos-setup.html

Something like:

```
mkdir -p ~/esp
cd ~/esp
git clone --recursive https://github.com/espressif/esp-idf.git
cd ~/esp/esp-idf
./install.sh esp32
. $HOME/esp/esp-idf/export.sh
```

Another option is to run everything inside `espressif/idf` docker image and install your tools there.

xobs, 2022-10-24:
Thanks. I did end up following your advice and now install ESP-IDF as part of every build: https://github.com/farpatch/farpatch/blob/main/.github/workflows/build-nightly-blackmagic.yml#L52-L55

This does add time to the build. It is a bit surprising that `esp-idf` has no access to any of the other steps. Perhaps that should be pointed out in the build steps? Or perhaps it's just common knowledge that I wasn't aware of -- different build steps may not have access to the same environment.

Either way, I'll try to arrange future versions of my software so that ESP-IDF doesn't require any external components to build.

kumekay, 2022-10-24:
It depends on how the action is implemented. If a step uses an action that runs in a docker container, it won't have access to the runner host environment.