espressif/esp-idf-ci-action #25: Using path corrupts git describe

timoxd7, 2022-12-06:
## Problem
Using the path parameter will set the working directory for the docker container to this set path.

### Example file structure
- Project
  - .git
  - sub-folder
    - src
    - CMakeLists.txt
    - ...
  - ...

If now the sub-folder is used as "path" parameter, the .git folder will not be visible inside the docker container. This will lead to different behavior then running the "idf.py build" command directly, as in the normal herachie the "git describe --tags" command will also search in parent folders.

## Fix
Add a second parameter, to specify "working-dir" and "build-path" to do some kind of using the "working-dir" as volume for docker, but then run "cd build-path" inside the docker container. Don't know how to simply do this inside this extension, but might need to update the docker image as well.

kumekay, 2022-12-06:
@timoxd7 Thank you for the issue, 
as a workaround, you can unset `path` parameter (mount whole repo) and add a custom command with CWD

```
jobs:
  build:
    ...
    - name: esp-idf build
      uses: espressif/esp-idf-ci-action@v1
      with:
        command: idf.py -C ./sub-folder build 
```

kumekay, 2022-12-06:
@timoxd7  please also take into account that `actions/checkout@v2` by default creates shallow clones, so you may want to pass `fetch-depth: '0'` to checkout step

timoxd7, 2022-12-06:
Yes, already did this. I tried it to run now, but id didn't work. I took the command you mentioned, but still can't access the tag from git... If i install the idf entirly inside the VM instead of the docker image, then it woks...

kumekay, 2022-12-07:
@timoxd7 

I experimented, and likely it's due to building directory ownership. https://github.com/kumekay/test_idf_action/actions/runs/3637431706/jobs/6138436795#step:3:76

Could you please try to use this test branch of the action?

```
uses: espressif/esp-idf-ci-action@bugfix/git_inside_repo
```

timoxd7, 2022-12-07:
@kumekay Thank you, i just tested it but now, in combination with `command: idf.py -C ./remote build` it will produce this output, after running the `export.sh` inside the docker image: 
```
Go to the project directory and run:

  idf.py build

/home/runner/work/_temp/a31f9873-05a3-4f2f-a314-abbd62ac3a4f.sh: line 4: idf.py: command not found
```

I built my own docker image, as i need a modifyed version of idf. It is, however, a docker image with exactly the same Dockerfile and entrypoint.sh as inside stock idf.

kumekay, 2022-12-24:
@timoxd7 I'm sorry for the long answer. Could you please try the branch `uses: espressif/esp-idf-ci-action@bugfix/git_inside_repo`

There was a problem with arguments escaping. 

An example with passing job: https://github.com/kumekay/test_idf_action/actions/runs/3771815462/jobs/6412477832

kumekay, 2023-01-03:
@timoxd7 !27 is merged now, and git should work inside the job. 
Feel free to reopen this issue in case of any problems, and thank you for contributing!

closed 2023-01-03 by kumekay