espressif/esp-idf-ci-action #28: A lot of blank lines when running build

enricobertolotti, 2023-01-09:
Is there a reason there are so many blank lines when running this action? All in all this action outputs around ~1k lines, most of them blank. It makes it hard to inspect the output

[image]

### Setup:
Self-hosted on ubuntu (rpi)

### Workflow:
```yaml
jobs:
  build:
    steps:
      - name: ESP-IDF Build
        uses: espressif/esp-idf-ci-action@v1
        with:
          esp_idf_version: v5.0
          target: esp32s2
          path: '.'
```

kumekay, 2023-01-09:
Hello @enricobertolotti
It looks like the build output is not visible at all. Are you able to read the logs of ninja?

kumekay, 2023-01-09:
@enricobertolotti

I was able to reproduce the issue, probably the root cause of this is error hints system added in ESP-IDF 5.0,

The problem present with IDF 5.0 https://github.com/kumekay/ci_playground/actions/runs/3877593140/jobs/6612783850

But it works fine with IDF 4.4
https://github.com/kumekay/ci_playground/actions/runs/3877611222/jobs/6612821848

tajnymag, 2023-10-24:
Can confirm. Still happening on the current version of the action and IDF 5.0.2

fhrbata, 2024-01-15:
Hello,

I tried it and it seems this is presented in v5.0 only. `fail` - empty lines instead of cmakes progress output.

- v4.4 - ok
- v5.0 - fail
- v5.1 - ok
- v5.2-beta2 - ok
- latest - ok

I will take a look what is missing in v5.0. Thank you

fhrbata, 2024-01-15:
Hello,

this should be already fixed by https://github.com/espressif/esp-idf/commit/8be7a054871cd041a242a9aa8f7e83e9a7a28ee1. The problem seems to be that `v5.0` docker image does not contain this fix, because it was last updated Dec 2, 2022. I guess the major release image is not updated once released. I tried `v5.0.5` and it seems to be working as expected. You can also try to use `release-v5.0` with latest `v5.0` fixes.

@kumekay  I believe this can be closed.

Thank you

kumekay, 2024-01-18:
@fhrbata Thank you for taking a look

closed 2024-01-18 by kumekay