espressif/esp-idf-ci-action #29: Working Dir Not found

sylwek845, 2023-04-11:
compilation fails due dir not found, my structure
lets say the company name is "company"

projectA
    - components
        - a
        - b
    - main

Inside "a" I have CMakeLists.txt with some file lets say "main2.c"
when I compile it fails because it cannot find dir "/app/company/ProjectA/components/a/main2.c"

Because the file is actually in "./ProjectA/components/a/main2.c"

it seems like the docker current dir (.) is "/app/company" which causes problems.

Thanks

kumekay, 2023-04-12:
@sylwek845 

Could you please provide more details on the configuration of your CI workflow?
Is `ProjectA` a name of the repository or a directory inside the repo? If it's a directory inside the repository, did you set the `path` input to `ProjectA`? Like:

```
jobs:
  build:
    ...
    steps:
    - ...
    - uses: espressif/esp-idf-ci-action@v1
      with:
        path: 'ProjectA'
```

sylwek845, 2023-04-18:
Closing. It was the config, had double .c file declaration which caused the issue. Thanks

closed 2023-04-18 by sylwek845