espressif/esp-idf-ci-action #34: Unable to give custom path to build

usmanmehmoodSSP, 2023-08-31:
I am checking out my code to a specific folder and when I give the same folder to the action's path, it appends some stuff to it.

This is my action.yaml
```yaml
- name: Checkout Application Code
  uses: actions/checkout@v3
  with:
    path: ${{ github.workspace }}/application

- name: esp-idf build
  uses: espressif/esp-idf-ci-action@v1
  with:
    esp_idf_version: v5.1
    target: esp32c3
    path: ${{ github.workspace }}/application

- name: Upload build
  uses: actions/upload-artifact@v3
  with:
    name: FirmwareFiles
    path: ${{ github.workspace }}/application/build/app_name.bin
```

And during the build process it's not able to find it and says this

```shell
CMakeLists.txt not found in project directory /app/my_username/app_name/home/runner/work/app_name/app_name/application
```

So as I see it, it is appending `/app/my_username/app_name` to `${{ github.workspace }}/application`

kumekay, 2023-08-31:
@usmanmehmoodSSP 

Please try to specify the path in the build step relative to GitHub workspace:

```
    path: application
```

usmanmehmoodSSP, 2023-08-31:
@kumekay it works perfectly with this. Thanks a bunch 😄

closed 2023-08-31 by usmanmehmoodSSP

kumekay, 2023-08-31:
@usmanmehmoodSSP Great, I added a note to the README https://github.com/espressif/esp-idf-ci-action/pull/35

epneo-banuprakash, 2024-04-01:
Hi ,
   I am also facing the same issue , i changed the path as specified above but it seems not working for me 
[image]

kumekay, 2024-04-08:
@epneo-banuprakash please try to remove `${{ github.workspace }}` from the build action, it expects a relative path. I.e. the last line on the screenshot:

```
    path: application
```

I see you use `actions/checkout` twice. Is it intended?