espressif/esp-idf-ci-action #36: question regarding the esp_idf_version

bhuvanchandra, 2023-09-29:
In regards to the versioning, it is always confusing. The docker-tagged versions are not proper, the versions are random with no proper reference to the release of esp-idf, which makes it really hard to rely on the docker image tag. If anyone knows how the version is supposed to be interpreted with esp-idf and the docker tags please do let me know. I'm in the stage to drop using the docker image and make a custom one.

kumekay, 2023-09-29:
Hello @bhuvanchandra 
this action is a thin convenience wrapper around official ESP-IDF docker image: https://hub.docker.com/r/espressif/idf

The image is versioned reliably with tags:

- latest: tracks master branch of ESP-IDF
- vX.Y: corresponds to ESP-IDF release vX.Y
- release-vX.Y: tracks release/vX.Y branch of ESP-IDF

You can specify the IDF image tag using `esp_idf_version` parameter in your workflow.

If version is not specified, the latest one will be used. 

With which particular tag do you have unexpected results?

bhuvanchandra, 2023-09-29:
@kumekay
Thank you for the clarification. I understand now. My confusion likely arose because the Docker images of older versions are being updated, which led me to believe that the versions aren't consistent. Other than that confusion, I haven't encountered any specific issues. Since I'm using the 5.1.1 stable version, I used the release-v5.1 Docker tag.

closed 2023-09-29 by bhuvanchandra

kumekay, 2023-09-29:
@bhuvanchandra Right, release-vX.Y are moving, but you can use the tag `v5.1.1` https://hub.docker.com/layers/espressif/idf/v5.1.1/images/sha256-5e8e9e02a7f41ca549a03e0df4ab48d1b9f7aaabf040797a5963e8dc548a8e52?context=explore