espressif/esp-idf-ci-action #37: Add optional esp-idf path

texruska, 2023-11-09:
Hello,

I use a modified version of esp-idf. I would like to specify the path to it instead of using the tagged release

kumekay, 2023-11-11:
Hello @texruska, thank you for opening the issue.

This action is a thin wrapper around the official ESP-IDF docker image: https://hub.docker.com/r/espressif/idf

This docker image contains downloaded tools and python environment for the specific commit of ESP-IDF. While it's technically possible to add the custom IDF_PATH, you will still need to specify IDF version and there is no warranty that the environment will work. 

For the case of modified IDF, I would not suggest using this action, but instead build the ESP-IDF docker image directly in your GitHub workflow.

May I also ask what is the main reason for using a patched version of ESP-IDF?

texruska, 2023-11-13:
I have a few changes to my own fork that haven't made it to upstream yet 

In the meantime I've written my own github actions without using docker. The main benefit of using a docker image would be to reduce the number of cpu minutes used/make it quicker when running the action?

This can be closed, I see that the way this action is architected would make it hard to implement and my alternative works until my patches make it into the idf

kumekay, 2023-11-13:
@texruska You may try this branch: https://github.com/kumekay/esp-idf-ci-action/tree/idf-path
but this is a quirky solution which I would prefer not merge it to the official action.

closed 2023-11-13 by kumekay