espressif/esp-idf-ci-action #38: Difficulty Accessing Private GitHub Repo with IDF Component Manager in CI Pipeline

hayschan, 2024-01-10:
Hello,

I'm encountering an issue with `espressif/esp-idf-ci-action@v1` while trying to build a project in my CI pipeline. The project uses the IDF Component Manager to fetch a component from a private GitHub repository.

Despite several attempts, I'm unable to grant access to the private repository for the IDF Component Manager running on GitHub. I've tried methods similar to those discussed in [this forum post](https://esp32.com/viewtopic.php?f=13&p=125861), including using ssh-agent and configuring known_hosts, but to no avail.

Could you please clarify:
1. Is `espressif/esp-idf-ci-action` designed to support accessing private repositories through the IDF Component Manager using `idf_component.yml`?
    ```yaml
    ## IDF Component Manager Manifest File
    dependencies:
      espressif/button: "^2.5.0"
      ## Required IDF version
      idf:
        version: ">=4.1.0"
    
      # For components in git repository:
      esp_arithmetic:
        path: "./component_add"
        version: "*"
        git: "git@github.com:kaspernyhus/multi_component_test.git"
    ```
2. If supported, could you provide guidance or examples on how to configure this correctly?

kumekay, 2024-01-12:
Hello @hayschan and thank you for raising the topic.

Access to private repositories wasn't in our scope yet. At the same time, we can make it work. 

The flow is different for SSH based repository access and over HTTPS. Is it crucial for you to use SSH access, or can you switch to HTTPS?

In this case, we can support passing of arbitrary environment variables, that can be used for authentication.

hayschan, 2024-01-13:
As long as the IDF component manager can access private GitHub Repo, I am fine with any solution.

Another workaround might be having private component on the ESP registry. These component can be grabbed from the ESP registry, but not available to public unless the user have a special authentication. The authentication can be password, or a pair of private and public key.

kumekay, 2024-04-29:
@hayschan I apologize, didn't notice your response. 

To access a private repo through HTTPs you can use environment variables in the manifest for credentials https://docs.espressif.com/projects/idf-component-manager/en/latest/reference/manifest_file.html#environment-variables-in-manifest

And #43 adds a way to pass custom arguments, including environment variables.

kumekay, 2024-11-01:
@hayschan Now (in v1.2.0 of this action) it's possible to pass an env variable with credentials and use environment variables in the manifest.

I would like to close this issue, if you have any ideas, or it doesn't work in your case, feel free to reopen it.

closed 2024-11-01 by kumekay