espressif/esp-idf-ci-action #45: Target multiple paths

nopnop2002, 2024-06-08:
I have multiple projects that use one component.
```
project
 - component
 - path1
 - path2
 - path3
```

How can I target multiple paths?

I tried the following, but an error occurred.

```
      matrix:
        idf_version:
          - release-v4.4
          - release-v5.0
          - release-v5.1
        idf_target:
          - esp32
          - esp32s2
        idf_path:
          - path1
          - path2
          - path3

    runs-on: ubuntu-latest
    steps:
    - name: Checkout repo
      uses: actions/checkout@v4
      with:
        submodules: 'recursive'
    - name: Build Application with ESP-IDF
      uses: espressif/esp-idf-ci-action@v1
      with:
        esp_idf_version: ${{ matrix.idf_version }}
        target: ${{ matrix.idf_target }}
        path: ${{ matrix.idf_path }}
```

kumekay, 2024-06-10:
@nopnop2002 I cannot find any issues in your config just by looking at it.
What kind of error has occurred? Please share the logs.

nopnop2002, 2024-06-10:
@kumekay 

I used this repository for test.
https://github.com/nopnop2002/esp-idf-DFPlayerMini

This is my espidf-compile.yml:

```
name: esp-idf-ci-action

on:
  pull_request:
    paths:
      - "main/**"
      - "components/**"

  push:
    paths:
      - "main/**"
      - "components/**"

  workflow_dispatch:
  repository_dispatch:

jobs:
  build:
    strategy:
      fail-fast: false
      matrix:
        idf_version:
          - release-v4.4
          - release-v5.2
        idf_target:
          - esp32
          - esp32s2
          - esp32c3
          - esp32s3
        idf_path:
          - GetStarted
          - PlayWithGpio
          - PlayWithCron

    runs-on: ubuntu-latest
    steps:
    - name: Checkout repo
      uses: actions/checkout@v4
      with:
        submodules: 'recursive'
    - name: Build Application with ESP-IDF
      uses: espressif/esp-idf-ci-action@v1
      with:
        esp_idf_version: ${{ matrix.idf_version }}
        target: ${{ matrix.idf_target }}
        path: ${{ matrix.idf_path }}
```

This is log:

[image]

malachib, 2024-12-21:
Hi, I am hitting something similar.  On mine it feels like it's not cleaning out the build folder and getting confused on targets that way.  Not sure though.