espressif/esp-idf-ci-action #48: Sharp Corner: Easy to select wrong `esp_idf_version`

BrianPugh, 2025-02-16:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [x] Provided a clear description of your suggestion.
- [x] Included any relevant context or examples.

### Issue or Suggestion Description

A bit of a sharp corner, a github action definition might look like:

```yaml
      matrix:
        idf_ver: ["v5.0", "v5.1", "v5.2", "v5.3", "v5.4", "latest"]
        idf_target: ["esp32"]
        include:
          - idf_ver: "v5.2"
            idf_target: esp32s2
          - idf_ver: "v5.2"
            idf_target: esp32c3
          - idf_ver: "v5.2"
            idf_target: esp32s3
      steps:
          - name: esp-idf build library
            uses: espressif/esp-idf-ci-action@main
            with:
              esp_idf_version: ${{ matrix.idf_ver }}
              target: ${{ matrix.idf_target }}
```

This looks perfectly reasonable. However, it doesn't actually work _exactly_ as expected. One would expect the esp_idf_version `v5.0` to use the latest `v5.0.X` patch release of esp-idf. However, this is passed along as a docker tag, and `v5.0` is a pretty old docker tag that contains the initial release `v5.0.0`. The same goes for all the other versions. What actually work is:

```yaml
              esp_idf_version: release-${{ matrix.idf_ver }}
```

But this isn't exactly correct either, as this contains all the latest changes that might not actually be in a tagged version.

I think the real misleading issue here, is that the docker tag `v5.0` (and similar) should **lead to the latest patch release of that version**. I don't think anyone using the tag `v5.0` is/should rely/expect to get `v5.0.0`. They would expect to get the latest patch release.

BrianPugh, 2025-02-16:
Discovered while debugging CI for https://github.com/joltwallet/esp_littlefs/pull/221, because the `littlefs` partition subtype only exists in later patch versions.

kumekay, 2025-02-17:
Hi @BrianPugh 

I understand your confusion, the current structure of docker image tags follows the structure of git tags https://github.com/espressif/esp-idf/tags, and we do not maintain any rolling tags there. And `v5.0` is the tag for the first stable release of 5.0 and there is no `v5.0.0` tag at all. This behavior has its historical roots.

I believe we can add (and document) a rolling tag, like `stable-v5.3` because the existing `release-v5.3` points to the branch, not to the release. However, this is out of the scope of this repo and must be addressed in the https://github.com/espressif/esp-idf directly.